RUB-SysSec/DroneSecurity #2: Local Reproduction/ CA - YOW region - DJI Air 2s - OcuSync 3.0

obriensystems, 2023-03-03:
Good Morning, thank you for the excellent article and associated repo for capturing droneID radio traffic.
I was also under the impression that DroneID was encrypted.  There was a POC last year in Ottawa that tracked a range of 40KM from YOW.  I didn't realize the 

OcuSync 2.0 to 3.0
I wish to contribute to your project first by cloning your repo and reproducing your base setup towards the goal of automated tracking of various drones starting with my DJI Air 2S with a mini 2 as a backup.  If required I will move up to the Mavik 3.

I currently fly the drone in Transport Canada approved airspace under the VLOS flight certificate and would like to combine your software/hardware setup eventually with AI based visual tracking.

Background: found your repo and paper via the Wired Magazine article https://www.wired.com/story/dji-droneid-operator-location-hacker-tool/

I will leave project reproduction and status on your repo as I go - in this issue id - with your permission or on my fork.

# Work Items

## WI 1: 20230302: SDN selection
The purchase of the SDN radio is a bit more expensive that the first drone itself so I would like to verify the recommended model.
On your readme the model is https://github.com/RUB-SysSec/DroneSecurity#drone-id-receiver-for-dji-ocusync-20
"Ettus USRP B205-mini"

On your paper https://www.ndss-symposium.org/wp-content/uploads/2023/02/ndss2023_f217_paper.pdf the model is a USRP B200mini
"Our setup uses a USRP B200mini SDR that we connect to a laptop"

I assume the following model is supported and will purchase
https://www.ettus.com/all-products/usrp-b205mini-i/

n0vichkov, 2023-03-23:
Hi, we are trying to reproduce the same and after our investigations we got an answer that proposed decoding method does not work with DJI Air 2S with OsuSync 3.0 (we have tested live receiver (with some changes for USRP X310 as RF) and offline decoder. I suppose, that modulation and decoding method differs

maxx, 2023-04-30:
@fmichaelobrien @n0vichkov looking forward to hearing how it went with occusync 2.0! please update if you were successful. I haven't yet seen a successful reproduction and am waiting for someone to confirm before I dive in.

obriensystems, 2023-05-01:
Still at step 1: new to SDR - setting up my B205mini using a VMware Ubuntu VM on one of older Mac's (intel chip)
Following Whitney's tutorials
https://www.hackster.io/whitney-knitter/getting-started-with-the-ettus-b205mini-in-gnu-radio-e0d3ea
https://www.hackster.io/whitney-knitter/basic-rf-test-verification-on-the-b205mini-with-gnu-radio-1cd612

[image]

aholtzma-am, 2023-09-08:
If you post a baseband recording of Occusync 3.0, I can take a look to see what the differences are wrt 2.0.

Vlad71527, 2023-10-15:
https://drive.google.com/file/d/1tTH773umwQrek_QaHCVn9fYpJtGBJ3UH/view?usp=sharing

tmbinc, 2023-12-03:
@Vlad71527 - unfortunately your images don't work anymore here (404 from github) 

The first one looked like a regular data packet, with a (variable; from a certain set) ZC symbol at the beginning, in the middle, and at the end.

The second one looked like a DroneID packet, with two (fixed - always 600 and 147) ZC symbols in the middle (around a data symbol). The latter should decode just fine. Do you have timestamps for within your capture?

So far I have not seen any OcuSync version not using DroneID packets in the same format. (The only difference seems to be whether the empty prefix symbols are there or not.) Difficulties to decode seem mostly from the fact that the somewhat basic synchronization algorithm used here requires a very good quality signal.

gettyhub, 2024-05-11:
Is there any way to make this work with antsdr or bladeRF?

Skeletoskull, 2024-05-30:
I am using X310 for this, has someone used it for this project, if yes what I should change to make the code work for my x310