RUB-SysSec/DroneSecurity #4: DroneID packet frequencies

Cyclone699, 2023-03-05:
Hello, 

I have to ask - the freqs of the droneIDs that uses in your code - did you check them? did you really see droneIDs in ALL this freqs?
I read the droneID article - "DJI drone IDs are not encrypted" and the work from github - https://github.com/proto17/dji_droneid,
and never saw your freqs.

moreover - when I tried to find manually droneIDs in my missing freqs - I didn't find any.
so if you tested that - I would like to know.

Something else - Should droneID works with other DJI drones? like DJI Lightbridge?

I would appreciate a response on these issues.

mrlnc, 2023-03-06:
Good point, the frequencies in our code might actually be wrong, it wasn't our focus. It's great you double-checked with [proto17/dji_droneid](https://github.com/proto17/dji_droneid).

Our code is OcuSync-specific, so it cannot receive Lightbridge or the new Bluetooth/WiFi-based Remote-ID.

I'll include that in the next FAQ update.

jhonnybonny, 2023-03-07:
> Good point, the frequencies in our code might actually be wrong, it wasn't our focus. It's great you double-checked with [proto17/dji_droneid](https://github.com/proto17/dji_droneid).
>
> Our code is [...]

I want to ask if a half duplex is suitable
or you only need a fulduplex device

jhonnybonny, 2023-03-07:
Like hackRF ?

mrlnc, 2023-03-07:
The code is RX-only, it cannot send but only receive. So as long as you can receive with the required bandwidth, it should work. Please note we didn't test HackRF. 

With this issue (potentially wrong channel list) in mind, you could always record samples manually and then use the offline receiver.

jhonnybonny, 2023-03-08:
Great 
Well then, if it doesn’t start, I’ll rewrite it under HackRF

closed 2023-03-08 by nschllr