RUB-SysSec/DroneSecurity #7: List of minimum SDR requirements

catosh, 2023-03-07:
A list of minimum SDR hardware requirements would be nice. 
I have a NESDR 5, before playing with tools IDK if the frequencies to listen for are in the range of my device or not.

Skotis, 2023-06-14:
Not all the freqs but those Iv seen

    2.4595 GHz
    2.4445 GHz
    2.4295 GHz
    2.4145 GHz
    2.3995 GHz
    5.7565 GHz
    5.7765 GHz
    5.7965 GHz

You also need quite a bit of bandwith to make it work. I have no number here but the protocol as a whole uses 20 MHz ish at a time

nschllr, 2023-06-14:
Thanks for that list! 

To be able to detect the signal, the SDR should be capable of capturing on both 2.4GHz and 5.8GHz frequencies. If your SDR only supports 2.4GHz, you can use the companion app to force the drone to operate on the lower channel.

According to the bandwidth:
We tested it with 20MHz, and it worked in our case. Since the DroneID signal itself occupies 15.36MHz, for example, a HackRF is capable of receiving the signal.

closed 2023-06-14 by nschllr