RUB-SysSec/DroneSecurity #8: Does it require a high signal-to-noise ratio for USRP to extract the signal?

LeTBr, 2023-03-08:
I happen to have a USRP B200 SDR which works for your code. But it can not capture signals of my DJI Air 2S. Does it require a high SNR to extract the signal. How did you configure the parameters of your B200, or do you use a high gain antenna to get the signals and decode the drone ID? I am very curious. In my case, the code will never be touched for unqualified bw.
[image]

revlis01, 2023-03-10:
@LeTBr Please note that Air 2S has the OcuSync 3.0 protocol and it might not work as the project is tested on OcuSync 2.0 devices as stated in the paper.

Also experimenting with B200 and getting packet detections, but it seems that an average i7 pc can't process the data. To avoid overflow issues at 50MSPS, the PC will have to be very powerful and has to have USB 3.0 ports.

KarlL2, 2023-03-30:
https://github.com/RUB-SysSec/DroneSecurity/issues/2#issuecomment-1480795529 seems to confirm that the current code does not work with DJI Air 2S which uses OsuSync 3.0.

closed 2023-04-11 by LeTBr