RUB-SysSec/DroneSecurity #23: Decoding failed for DJI Mavic Air 2

waseem-arshad94, 2023-12-22:
Dear Sir;

I recorded the DJI MAVIC AIR 2 signal inside shielding room using adrv9361z7035 SDR board. I recorded the 3 Scenarios, one just ON (Mavic_air_2s_1000.fc32) and 2nd one turn on the propellers (Mavic_air_2s_1100.fc32) and 3rd take off (Mavic_air_2s_1110.fc32)and fly at the distance of 1m from the receiver. Whenever I run the offline receiver but decoding failed.I am sharing the complete log file and PNG files of in-spectrum testing. Kindly help needed. Please find the attachment.

[DJI_MAVIC_AIR_2.zip](https://github.com/RUB-SysSec/DroneSecurity/files/13752723/DJI_MAVIC_AIR_2.zip)

Thanks in advance

nschllr, 2024-01-05:
If the screenshots show your entire recording, then no droneID packets are captured. Additionally, I believe that the sampling rate is incorrect; the captured packets appear to be the downlink, which should be 20MHz wide (as far as I know for that drone type).

closed 2024-01-05 by nschllr

waseem-arshad94, 2024-01-05:
I set a sampling rate of 15.36 MHz.