RUB-SysSec/DroneSecurity #25: DroneID Packet duration

SpongeBobaa, 2024-01-03:
Regarding the experimental results of my DJI Mavic Air 2, I continuously monitored the spectrum for more than ten minutes and did not seem to find a suitable DroneID burst signal. I have some thoughts below.

In your paper, it was mentioned that the DroneID data packet is broadcasted every 640ms, but the actual data frame is only 648 μs. Despite its high transmit frequency and bandwidth of 15.36MHz, it is approximately 600 μs of the  duration makes it difficult to detect such signals in the spectrum. Is my analysis correct?

I also tried to record signals within a certain frequency band, such as (5755MHz-5805MHz), with a sampling bandwidth of 50M, but I was unable to successfully detect the corresponding DroneID signal.

Do you have any good suggestions?

nschllr, 2024-01-05:
Hi, 
> Despite its high transmit frequency and bandwidth of 15.36MHz, it is approximately 600 μs of the duration makes it difficult to detect such signals in the spectrum. Is my analysis correct?

Correct, if you look manually into such a recording it might be hard to spot one of the packets if they are sent on the same band as the downlink.  if you have a receiver which cannot record the entire band (e.g. 2.4Ghz) the packets might also are on an other band outside your recorded frequency.

You can try to set a fixed frequency range in the settings of the drone, so that it only uses the 2.4GHz band.

SpongeBobaa, 2024-01-09:
Thanks for your reply

From what I have learned, the burst signal of DroneID is not a fixed frequency, it may be constantly changing, and the working channel of DroneID signal is separate from the image transmission and remote control channels. I can fix the transmission frequency of the drone, but how do I fix the frequency of the DroneID signal?

I did not detect the signal, so I am unable to determine the frequency of the DroneID burst signal.

SpongeBobaa, 2024-01-09:
Will the DroneID signal be sent as soon as the drone starts running? Whether indoors or outdoors.

SpongeBobaa, 2024-01-12:
I think I should have found some burst signals for DroneID. Is the data frame in the following graph correct?

[image]

hechunzhe, 2025-03-21:
Hello, I would like to ask what adjustments you made to capture the DroneID frames. I have been trying to collect them recently, but after recording for twenty minutes, I still couldn’t capture anything. My sampling rate is set to 30.72e6. I look forward to your response!