RUB-SysSec/DroneSecurity #30: Unable to Lock into Mavic 2 (Ocusync 2) Drone Frequency

BryanCastorius, 2024-07-03:
Dear @mrlnc,

My current setup uses USRP B-205 mini connected to a 2.4/5.8 GHz antenna for the RX2 to decode DJI Mavic 2 which uses Ocusync 2.0 protocol. I tried running the code both in indoor and outdoor environment and encounter the same issue, which it still stuck at the endless loop of trying to find the center frequency.
[image]
As I read through the code, I noticed that the list of frequencies are hard coded, is there a way to obtain the drone's actual frequency that it is communicating so that it can lock to it.

Furthermore, I have tried setting the 'Legacy = True' and still obtain the same problem.

Anirudhx234, 2024-07-03:
Hi, sorry that this is a bit unrelated, but were you able to get decoded payloads using the offline receiver? I'm currently trying to do it with a different SDR but have had no luck

BryanCastorius, 2024-07-04:
@Anirudhx234 

Yup, the offline receiver works perfectly fine for mine.

Anirudhx234, 2024-07-04:
Ok, good to know. I assume you ran the offline receiver in legacy mode since you're using the Mavic 2 Pro? I get the following error when I run it in legacy mode, so I was wondering if you made any changes to the original code. It seems to be locating the packet correctly based on the spectrum but I'm stuck here...
[image]

23miruthulaa, 2024-08-21:
Hi,I am having this overflow center frequency issue . so that I can't receive the drone ID payload details for ocusync 3.0(mavic 3 pro).Any solutions please suggest me

kiran809576, 2024-11-20:
Hi @BryanCastorius,  
I'm currently working on the same project, and the offline receiver code is working well. However, when I run the live receiver code, the data is being captured and processed, but no drone-id packets are being found.  
Were you able to resolve this issue with the live receiver? I’d appreciate any guidance or insights you can share.  

Thanks!

OP-drone, 2025-01-19:
Hi @kiran809576  
AntSDR E200 starts but does not find any droneid packages, I try on mavic mini 2 (Ocusync 2).
How did you manage to get AntSDR work with droneid_receiver_live ?